sabamdarif/termux-desktop #115: Some logs were accidentally hidden

David-xian66, 2025-04-26:
Original output：
```
╭─ [root  localhost] [~]
╰─❯ tx11start
declare -x ANDROID_ART_ROOT="/apex/com.android.art"
declare -x ANDROID_ASSETS="/system/app"
declare -x ANDROID_DATA="/data"
...（Omit）...
declare -x ZSH_CACHE_DIR="/data/data/com.termux/files/home/.cache/zinit"
declare -x vblank_mode="0"
```

```
╭─ [root  localhost] [~]
╰─❯ tx11start --debug
Failure: Module initialization failed
declare -x ANDROID_ART_ROOT="/apex/com.android.art"
declare -x ANDROID_ASSETS="/system/app"
declare -x ANDROID_DATA="/data"
...（Omit）...
declare -x ZSH_CACHE_DIR="/data/data/com.termux/files/home/.cache/zinit"
declare -x vblank_mode="0"
Termux:X11 application is not found.
You should choose latest release here:
 https://github.com/termux/termux-x11/releases/tag/nightly
After this you should download "termux-x11-arm64-v8a-debug.apk" or "termux-x11-universal-debug.apk" and install it.

Starting: Intent { cmp=com.termux.x11/.MainActivity }
Error type 3
Error: Activity class {com.termux.x11/com.termux.x11.MainActivity} does not exist.
xfce4-session: Cannot open display: .
Type 'xfce4-session --help' for usage.
```

```You should choose latest release here:
 https://github.com/termux/termux-x11/releases/tag/nightly
After this you should download "termux-x11-arm64-v8a-debug.apk" or "termux-x11-universal-debug.apk" and install it.

Starting: Intent { cmp=com.termux.x11/.MainActivity }
Error type 3
Error: Activity class {com.termux.x11/com.termux.x11.MainActivity} does not exist.
xfce4-session: Cannot open display: .
Type 'xfce4-session --help' for usage.
```
This section will only appear after I open debug, which is wrong and should be treated as error

sabamdarif, 2025-04-26:
No, it’s not accidental, I intentionally hid all logs in normal mode. Otherwise, running the tx11start command would clutter your terminal. That’s why I added a debug option, to help analyze things if something goes wrong.

David-xian66, 2025-04-27:
ok

closed 2025-04-27 by David-xian66