sabamdarif/termux-desktop #117: tx desktop - Gui Config and updater.

sb-codz, 2025-04-27:
Dear dev.
can you please make a gui re-config interface and update interface.

like your app store or other apps.
it's helpful.

sabamdarif, 2025-04-27:
I have a plan to create a gui desktop configuration app in the future, but currently, the app store needs a lot of work. Once that's done, I will start working on that the new app

sabamdarif, 2025-04-27:
not now but after few months or if everything goes right then after few weeks i will

sb-codz, 2025-04-30:
thanks.
it's helpful.

but currently required upgrade future
or version.

sabamdarif, 2025-05-01:
currently i have nothing like that but i will think about this

sabamdarif, 2025-06-10:
so i about your request, i tried and find that it best if you do them from terminal gui wrapper will just create issue . but i decide that i will add a theme & icon store to install them in just one click

sabamdarif, 2025-06-10:
[image]

this is what i will add, https://github.com/debasish-patra-1987/linuxthemestore, i ported it from termux

sabamdarif, 2025-06-15:
here is the theme store app:- https://github.com/sabamdarif/linuxthemestore/releases/tag/1.0.1-termux