sabamdarif/termux-desktop #120: Ubuntu Gui Problem

hacker50-us, 2025-04-28:
hello dev
I have a problem that ubuntu does not have gui like Native termux, so a black screen appears when activating the ```
gui --start
```
I hope the problem will be solved in the future

hacker50-us, 2025-04-28:
**Note:** I found out when I did an command 
```
sudo apt install xfce4 
```

to discover that it is not installed**_(I chose xfce4 when installing for the first time in termux )_**

sabamdarif, 2025-04-28:
"I think you misunderstood. This repo is about Termux native desktop, not a proot-distro-based desktop. You should read the README , I mentioned that proot-distro will be used to install various packages in Termux that are not available by default.

closed 2025-04-28 by sabamdarif