sabamdarif/termux-desktop #124: Audio not working right now.jow to fix it.

jim5690, 2025-05-10:


sabamdarif, 2025-05-10:
can you share me your device details, like manufacture, model android version

viphakorn, 2025-05-13:
i face same issues 

Lenovo Android 14 snapdragon 685

viphakorn, 2025-05-16:
I found solution

In `/data/data/com.termux/files/usr/bin/tx11start`

I replace with
```
pulseaudio --start --load="module-aaudio-sink"  --load="load-module module-native-protocol-tcp auth-ip-acl=127.0.0.1 auth-anonymous=1"  --exit-idle-time=-1

export PULSE_SERVER=127.0.0.1
```

`--load="module-aaudio-sink"` it's work for me

sabamdarif, 2025-05-16:
ok let me try

sabamdarif, 2025-05-16:
@viphakorn thanks it worked, i will update the script soon

Evslin0505, 2025-05-16:
Hello @viphakorn, I check your decision, it works, but  BUT when switching to the Bluetooth headset, the pulseaudio foll down. When I connect the headset and reboot Termux everything works, back to the speakers... pulse audio fall down again.

Katsuyamaki, 2025-05-31:
this fix is not working for me. I also tried to manually start pulseaudio server commands in termux and native xfce terminal. Any suggestions or troubleshooting tips?

commit 9b1ed4b5 (2025-05-16): fix #124 thanks @viphakorn, enable tab key to cycle through suggestion in zsh