sabamdarif/termux-desktop #127: Can't right click with physical trackpad

avoiceofreason, 2025-05-12:
I have a tablet with a physical keyboard/trackpad.

The keyboard works ok, the trackpad works ok for mouse movement and for left click, but does not do anything when I right click.

I can use the "Show mouse click help overlay" in Pointer settings and touch the right mouse button on screen which works ok, but can't find any setting to allow the physical trackpad to right click.

For info I don't use "tapping" on the trackpad, this is physical click on trackpad.

sabamdarif, 2025-05-12:
i don't own any pysical keyboard with trackpad so i can't help, sorry for that. but the last time when i try with a keyboard and a mouse it works fine for me

avoiceofreason, 2025-05-13:
Hi Saba,

You can also close this issue.

I found a setting in Android that had mysteriously been turned off that restored right click on the trackpad

Thanks

sabamdarif, 2025-05-13:
0k

closed 2025-05-13 by sabamdarif