sabamdarif/termux-desktop #135: [Q&A] How to update termux-desktop

ytoaa, 2025-06-06:
I want to update on a system that already has Termux-desktop installed. Of course, you can uninstall and reinstall, but that will delete other existing program settings, which will be inconvenient. Also, if you force a reinstallation, the settings will be deleted.

ytoaa, 2025-06-06:
> What do you want to update ?

For example, some have been patched due to recent sound issues, and I understand that the pdrun internal code has changed for today's update. Isn't there actually no other way to automate this except for forced reinstallation or reinstalling after initialization?

sabamdarif, 2025-06-06:
No unfortunately there is no to do that easily, it will be hard to add that kind of update option on a setup script

ytoaa, 2025-06-06:
> No unfortunately there is no to do that easily, it will be hard to add that kind of update option on a setup script

So, is there no way to automate it at the moment other than a full reinstallation or forced reinstallation?

sabamdarif, 2025-06-06:
see if you have no issue then you don't need to do anything. otherwise forced reinstallation will work