sabamdarif/termux-desktop #138: App Store not working

arun28082007, 2025-06-10:
[image]

[image]

it worked for 2 days but from Today's afternoon it stopped working suddenly

sabamdarif, 2025-06-10:
Go to update tab and click on check for update it will fix that or do 
rm -rf .appstore and relaunch the app store

arun28082007, 2025-06-10:
Nothing changed 😞.

sabamdarif, 2025-06-10:
i mean here run the command `appstore`

arun28082007, 2025-06-10:
Error: Missing files:
  • /data/data/com.termux/files/usr/share/applications/org.sabamdarif.termux.appstore.desktop
Please reinstall: appstore --install

then is run _appstore --install_ now it is working properly thanks