sabamdarif/termux-desktop #139: Module initialization failed

hasan1818666891, 2025-06-11:
╭─ [hasan localhost] [~]
╰─❯ tx11start --debug
Failure: Module initialization failed
Starting: Intent { cmp=com.termux.x11/.MainActivity }
ATTENTION: default value of option vblank_mode overridden by environment.
dlopen failed: cannot locate symbol "Xzs_Construct" referenced by "/system/lib64/libunwindstack.so"...dlopen failed: cannot locate symbol "Xzs_Construct" referenced by "/system/lib64/libunwindstack.so"...libGL error: failed to create dri screen
libGL error: failed to load driver: swrast

(xfce4-session:13180): libxfce4util-WARNING **: 07:55:32.355: Failed to get a ConsoleKit proxy: Could not connect: No such file or directory
** Message: 07:55:32.377: Failed to fetch _NET_NUMBER_OF_DESKTOPS; assuming 1
** Message: 07:55:32.377: Failed to get _NET_WORKAREA; using full screen dimensions
** Message: 07:55:32.378: Failed to fetch _NET_CURRENT_DESKTOP; assuming 0
** Message: 07:55:32.378: Failed to fetch _NET_CURRENT_DESKTOP; assuming 0
/data/data/com.termux/files/usr/bin/iceauth:  creating new authority file /data/data/com.termux/files/usr/tmp/ICEauthority
** Message: 07:55:33.020: Failed to fetch _NET_CURRENT_DESKTOP; assuming 0

(wrapper-2.0:13354): xfce4-battery-plugin-CRITICAL **: 07:55:36.107: battmon_dbus_monitor: assertion 'proxy != NULL' failed

hasan1818666891, 2025-06-11:
[image]

sabamdarif, 2025-06-11:
> dlopen failed: cannot locate symbol "Xzs_Construct" referenced by "/system/lib64/libunwindstack.so"...dlopen failed: cannot 

i don't know much but it looks like some system (os) specific issue

> libGL error: failed to load driver: swrast

also looks like some hardware-accelerated graphics related issue

hasan1818666891, 2025-06-11:
It was working correctly before, but I had to reinstall Termux for some reason, and now this is happening.

sabamdarif, 2025-06-11:
then it might be some termux side of issue, btw what issue do you have on the desktop ?

hasan1818666891, 2025-06-11:
Last question, 
What is your opinion, which termux version is good, 0.119.0-beta-3 or 0.118.3

sabamdarif, 2025-06-11:
latest stable