sabamdarif/termux-desktop #141: Neofetch segmentation fault

marcobibichan, 2025-06-23:
My termux have installed the termux desktop with hardware acceleration (Turnip Zink Vulkan). However, it returned "segmentation fault" when I try to use neofetch to fetch software info.

Device: Oneplus 13
Android version: 15
CPU: Snapdragon 8 Elite

[image]

sabamdarif, 2025-06-23:
it's actually fastfetch. 
for neofetch run 

```bash
pkg install neofetch -y
```
then

```bash
bash -c 'neofetch'
```
and see what happened

marcobibichan, 2025-07-07:
bash -c 'neofetch' works fine, but it shows cpu clockrate as 1.996Ghz ?
Should I type alias neofetch="bash -c 'neofetch'"  in .zshrc?

[image]

sabamdarif, 2025-07-07:
> btw is Snapdragon 8 Elite not supported by turnip?

yes i think it supports upto 7xx+

closed 2025-07-16 by sabamdarif