sabamdarif/termux-desktop #148: Hangover wine stuck on beginning

marcobibichan, 2025-07-07:
I installed hangover wine following the installtion instructions and typed "wine winecfg", "wine wineboot" and they displays nothing, ctrl C cannot stop.

Android 15
Oneplus 13
Snapdragon 8 Elite

  I have installed hangover on my another proot distro but encountered the same issue.

sabamdarif, 2025-07-07:
did you use the wine-hangover from the termux repo or anything else ?

marcobibichan, 2025-07-07:
> did you use the wine-hangover from the termux repo or anything else ?

Yes, I installed from your scipt.

I uninstalled hangover* (pkg uninstall hangover*) then reinstalled it (pkg install hangover*) then I typed "wine winecfg" then encountered another error:

[image]

closed 2025-07-16 by sabamdarif