sabamdarif/termux-desktop #149: There is a problem with the zed editor display

PuKYC, 2025-07-08:
I tried to install the zed editor in Distro Containers. This editor can only be installed using a script. After installation, I don't know how to add it to the menu. Also, the editor has problems displaying after startup.

[image]

[image]

PuKYC, 2025-07-08:
The same is true without using GPU
[image]

sabamdarif, 2025-07-08:
then it might be app's issue with termux environment, let me see

PuKYC, 2025-07-08:
> then it might be app's issue with termux environment, let me see

Is there any plan to add zed to the desktop installation list?🤔