sabamdarif/termux-desktop #153: Hardware acceleration stopped working

AnonN10, 2025-07-13:
installing `vulkan-icd-wrapper` with `vulkan(zink)` for everything results in what seems to be a missing ICD, therefore vulkaninfo, vkcube and vkmark fail with being unable to find VK_KHR_Surface instance extension

AnonN10, 2025-07-13:
fixed by installing [vulkan-wrapper-android_25.0.0-2](https://github.com/sabamdarif/termux-desktop/releases/tag/vulkan-wrapper-android_25.0.0-2) from releases manually (which are hidden now for some reason?)

sabamdarif, 2025-07-13:
That mean you must have have some issues during the setup process.

They are hidden because I didn't build those drivers I just upload them there so the setup script can download them easily

closed 2025-07-13 by sabamdarif