sabamdarif/termux-desktop #166: The following required not set

Bruh9980, 2025-08-17:
setup-termux-desktop -c autostart disable                [✓] Configuration variables loaded                         [-] Validating required configuration values...            [☓] The following required configuration values are not set:                                                            - user_name

Help me iam new idk how to fix that😭😭

mgueji, 2025-08-17:
> setup-termux-desktop -c autostart disable [✓] Configuration variables loaded [-] Validating required configuration values... [☓] The following required configuration values are not set: - user_name
> 
> Help me iam new idk how to fix that😭😭

You need to add your user name to the setup script:

`nano $PREFIX/etc/termux-desktop/configuration.conf` 

Then add in a new line at the bottom:

`user_name = YOURUSERNAME`

Bruh9980, 2025-08-17:
> You need to add your user name to the setup script:
>
> `nano $PREFIX/etc/termux-desktop/configuration.conf`
>
> Then add in a new line at the bottom:
>
> `user_name = YOURUSERNAME`

Oh thats work thanks😋👍🏻