sabamdarif/termux-desktop #167: I3 wm

Bruh9980, 2025-08-17:
I tried using i3 wm and installed terminal kitty but when I tried mod(alt)+enter it didn't show anything. Only the cursor changed to something weird.

[image]

Bruh9980, 2025-08-17:
> I tried using i3 wm and installed terminal kitty but when I tried mod(alt)+enter it didn't show anything. Only the cursor changed to something weird.
> 
> [image]

Likewise with rofi or nitrogen wallpaper engine. I don't know how to see the i3 wm running log in your project.

sabamdarif, 2025-08-17:
run `tx11start --debug` to see logs

Bruh9980, 2025-08-17:
> run `tx11start --debug` to see logs

Done i found the problem thanks.
Sorry to bother you today