sabamdarif/termux-desktop #178: tx11start then X11 blank screen

shyoo0000, 2025-08-30:
Hello, I had the problem. ​I ran the tx11start command, but all I'm getting is an X-shaped mouse cursor and a blank screen in X11. It's not working. I'm doing this on a Snapdragon 8 Gen 2 tablet with the 

> 3.mesa-zink-vulkan-icd-wrapper

 hardware acceleration option. Even adding _--legacy_ or _--nogpu_ after tx11start doesn't make it run properly. Why is this happening on the first installation and execution? I would be grateful for help.

```
System Info:
 System          :  Android 15
 Host            : 󰍹 samsung SM-X910
 Kernel          :  5.15.153
 CPU             : 󰍛 QTI SM8550 (8 vCPU)
 Architectures   : 󰻠 AARCH64
 Termux Version  :  0.118.3-Github
 Battery Level   :  85%
 Memory          : 󰘚 7.0Gi used, 10Gi total
 Temperature     :  56°C
Disk Usage:
 /storage/emulated                                                                                                                                                                                                                used 233G of 496G
[##################################################################################################################-----------------------------------------------------------------------------------------------------------------------------]
~ $ tx11start --legacy
/data/data/com.termux/files/usr/bin/tx11start: line 83: pulseaudio: command not found
/data/data/com.termux/files/usr/bin/tx11start: line 84: pacmd: command not found
declare -x ANDROID_ART_ROOT="/apex/com.android.art"
declare -x ANDROID_DATA="/data"
declare -x ANDROID_I18N_ROOT="/apex/com.android.i18n"
declare -x ANDROID_ROOT="/system"
declare -x ANDROID_TZDATA_ROOT="/apex/com.android.tzdata"
declare -x ANDROID__BUILD_VERSION_SDK="35"
declare -x BOOTCLASSPATH="/apex/com.android.art/javalib/core-oj.jar:/apex/com.android.art/javalib/core-libart.jar:/apex/com.android.art/javalib/okhttp.jar:/apex/com.android.art/javalib/bouncycastle.jar:/apex/com.android.art/javalib/apache-xml.jar:/system/framework/framework.jar:/system/framework/framework-graphics.jar:/system/framework/framework-location.jar:/system/framework/ext.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/knoxsdk.jar:/system/framework/framework-nfc.jar:/system/framework/tcmiface.jar:/system/framework/telephony-ext.jar:/system/framework/QPerformance.jar:/system/framework/UxPerformance.jar:/system/framework/esecomm.jar:/apex/com.android.i18n/javalib/core-icu4j.jar:/apex/com.android.adservices/javalib/framework-adservices.jar:/apex/com.android.adservices/javalib/framework-sdksandbox.jar:/apex/com.android.appsearch/javalib/framework-appsearch.jar:/apex/com.android.btservices/javalib/framework-bluetooth.jar:/apex/com.android.configinfrastructure/javalib/framework-configinfrastructure.jar:/apex/com.android.conscrypt/javalib/conscrypt.jar:/apex/com.android.devicelock/javalib/framework-devicelock.jar:/apex/com.android.healthfitness/javalib/framework-healthfitness.jar:/apex/com.android.ipsec/javalib/android.net.ipsec.ike.jar:/apex/com.android.media/javalib/updatable-media.jar:/apex/com.android.mediaprovider/javalib/framework-mediaprovider.jar:/apex/com.android.mediaprovider/javalib/framework-pdf.jar:/apex/com.android.mediaprovider/javalib/framework-pdf-v.jar:/apex/com.android.mediaprovider/javalib/framework-photopicker.jar:/apex/com.android.ondevicepersonalization/javalib/framework-ondevicepersonalization.jar:/apex/com.android.os.statsd/javalib/framework-statsd.jar:/apex/com.android.permission/javalib/framework-permission.jar:/apex/com.android.permission/javalib/framework-permission-s.jar:/apex/com.android.profiling/javalib/framework-profiling.jar:/apex/com.android.scheduling/javalib/framework-scheduling.jar:/apex/com.android.sdkext/javalib/framework-sdkextensions.jar:/apex/com.android.tethering/javalib/framework-connectivity.jar:/apex/com.android.tethering/javalib/framework-connectivity-t.jar:/apex/com.android.tethering/javalib/framework-tethering.jar:/apex/com.android.uwb/javalib/framework-uwb.jar:/apex/com.android.virt/javalib/framework-virtualization.jar:/apex/com.android.wifi/javalib/framework-wifi.jar:/apex/com.samsung.android.lifeguard/javalib/framework-lifeguard.jar:/apex/com.samsung.android.shell/javalib/framework-samsung-shell.jar"
declare -x COLORTERM="truecolor"
declare -x DEX2OATBOOTCLASSPATH="/apex/com.android.art/javalib/core-oj.jar:/apex/com.android.art/javalib/core-libart.jar:/apex/com.android.art/javalib/okhttp.jar:/apex/com.android.art/javalib/bouncycastle.jar:/apex/com.android.art/javalib/apache-xml.jar:/system/framework/framework.jar:/system/framework/framework-graphics.jar:/system/framework/framework-location.jar:/system/framework/ext.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/knoxsdk.jar:/system/framework/framework-nfc.jar:/system/framework/tcmiface.jar:/system/framework/telephony-ext.jar:/system/framework/QPerformance.jar:/system/framework/UxPerformance.jar:/system/framework/esecomm.jar:/apex/com.android.i18n/javalib/core-icu4j.jar"
declare -x DISPLAY=":0"
declare -x EXTERNAL_STORAGE="/sdcard"
declare -x HISTCONTROL="ignoreboth"
declare -x HOME="/data/data/com.termux/files/home"
declare -x LANG="en_US.UTF-8"
declare -x LD_PRELOAD="/data/data/com.termux/files/usr/lib/libtermux-exec-ld-preload.so"
declare -x MESA_NO_ERROR="1"
declare -x OLDPWD
declare -x PATH="/data/data/com.termux/files/usr/bin"
declare -x PREFIX="/data/data/com.termux/files/usr"
declare -x PWD="/data/data/com.termux/files/home"
declare -x SHELL="/data/data/com.termux/files/usr/bin/bash"
declare -x SHLVL="2"
declare -x TERM="xterm-256color"
declare -x TERMUX_API_VERSION="0.52.0"
declare -x TERMUX_APK_RELEASE="GITHUB"
declare -x TERMUX_APP_PID="4971"
declare -x TERMUX_APP__DATA_DIR="/data/user/0/com.termux"
declare -x TERMUX_APP__LEGACY_DATA_DIR="/data/data/com.termux"
declare -x TERMUX_APP__SE_FILE_CONTEXT="u:object_r:app_data_file:s0:c2,c258,c512,c768"
declare -x TERMUX_APP__SE_INFO="default:targetSdkVersion=28:complete"
declare -x TERMUX_IS_DEBUGGABLE_BUILD="1"
declare -x TERMUX_MAIN_PACKAGE_FORMAT="debian"
declare -x TERMUX_VERSION="0.118.3"
declare -x TERMUX__HOME="/data/data/com.termux/files/home"
declare -x TERMUX__PREFIX="/data/data/com.termux/files/usr"
declare -x TERMUX__ROOTFS_DIR="/data/data/com.termux/files"
declare -x TERMUX__SE_PROCESS_CONTEXT="u:r:untrusted_app_27:s0:c2,c258,c512,c768"
declare -x TMPDIR="/data/data/com.termux/files/usr/tmp"
declare -x XDG_RUNTIME_DIR="/data/data/com.termux/files/usr/tmp"
declare -x vblank_mode="0"
```

sabamdarif, 2025-09-01:
did you mean `mesa-zink-vulkan-icd-freedreno` ?

shyoo0000, 2025-09-01:
The problem was that nextDNS mistook the packets for advertisements and failed to install the necessary packages. Now it's fixed!

shyoo0000, 2025-09-01:
But I have another question. In the latest version, can't I select freedreno kgsl?

sabamdarif, 2025-09-01:
i am adding that wait few days https://github.com/sabamdarif/termux-desktop/tree/freedreno