sabamdarif/termux-desktop #189: Chroot install issues

kyurenbog, 2025-09-14:
I have the following errors.

[image]
[image]

kyurenbog, 2025-09-14:
I use this chroot https://github.com/Magisk-Modules-Alt-Repo/chroot-distro
As I understand it I need to use? https://github.com/sabamdarif/chroot-distro

sabamdarif, 2025-09-14:
Magisk-Modules-Alt-Repo/chroot-distro wouldn't work i did mention [here](https://github.com/sabamdarif/termux-desktop/blob/main/docs/proot-container.md#note--for-chroot-distro-flash-this-module-first--sabamdarifchroot-distro), that you need my chroot-distro module

kyurenbog, 2025-09-14:
> Magisk-Modules-Alt-Repo/chroot-distro wouldn't work i did mention [here](https://github.com/sabamdarif/termux-desktop/blob/main/docs/proot-container.md#note--for-chroot-distro-flash-this-module-first--sabamdarifchroot-distro), that you need my chroot-distro module

I installed yours.
Next is this error. What can you tell me about it?

[image]

kyurenbog, 2025-09-14:
It doesn't even ask what fonts to install.

kyurenbog, 2025-09-14:
Plus a question, after running vncstart, why do I get into the termux environment and not ubuntu

sabamdarif, 2025-09-14:
> > install jq in advance
> 
> There is an error No .tar.xz fonts found for

this is why i said you to install jq in advance, you need jq to make that list appear, jq is a program

kyurenbog, 2025-09-14:
Yes, please forgive me. I didn't quite understand everything correctly.

kyurenbog, 2025-09-14:
> this script doesn't install the desktop environment in any proot distro, it install them in termux , the proot-option is optional, i think you didn't read the readme check [this section](https://github.com/sabamdarif/termux-desktop#easily-install-termux-native-gui-desktop) and [this](https://github.com/sabamdarif/termux-desktop/blob/main/docs/proot-container.md) too

plus there was another error, but the acceleration works.
In your enable-hw-acceleration script you have TURNIP_MESA_VERSION="25.1.5", but you don't have one. I only see turnip-25.2.2-aarch64.zip
That's why there's an error like this in the photo.

[image]

kyurenbog, 2025-09-15:
No thanks