sabamdarif/termux-desktop #209: tx11start black screen, works with --no-gpu

Yuri-jeik-Becker, 2025-09-30:
Here is the terminal:
```shell
System Info:
 System          :   Android 15
 Host            : 󰍹  Redmi 2405CRPFDG
 Kernel          :   5.10.226
 CPU             : 󰍛  QTI SM7435 (8 vCPU)
 Architectures   : 󰻠  AARCH64
 Termux Version  :   0.119.0-beta.3-
 Battery Level   :   73%
 Memory          : 󰘚  4.2Gi used, 7.5Gi total
 Temperature     :   30°C

Disk Usage:
 /storage/emulated                                                                                                                                                 used 85G  of 238G
 [###############################################################-----------------------------------------------------------------------------------------------------------------]
~ $ tx11start
/data/data/com.termux/files/usr/bin/tx11start: line 154: 12290 Aborted                    env DISPLAY=:0 XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg dbus-launch --exit-with-session $de_startup > /dev/null 2>&1
~ $ tx11stop
Termux:X11 Stopped Successfully
``` 

I choose mesa-vulkan-icd-freedreno(turnip) for my Adreno 710. I really want to use turnip drivers instead of going the virgl route, plz help!!!

sabamdarif, 2025-09-30:
adreno 710 isn't officially supported by turnip, but there is a way and that patch has been added here https://github.com/termux/termux-packages/pull/26528/files#diff-31e4c22d60e491bd3b8cde9587e30670cde7799d768c90660a29dc1b69690fdd

either with for it to get merged in termux or download the build artifacts from here https://github.com/termux/termux-packages/actions/runs/18118686852/job/51559300166?pr=26528

Yuri-jeik-Becker, 2025-09-30:
Oh, ok... Can close issue

closed 2025-09-30 by sabamdarif